OllisGit/OctoPrint-CostEstimation #4: Cost not Displaying

jason-a-alexander, 2020-10-04:
For some reason I can not get cost to display.  I have version 3.0 installed.   I have done a removal and reinstall, I have turned off the connection to filament manager and still I don't get a cost.   It looks like the plugin is just not displaying.  I have also removed any plugins that modify the CSS like themify, octoprint-wide, and dashboard.  

[image]

[image]

Any thoughts?

OllisGit, 2020-10-07:
Hi @jason-a-alexander,

I released a new version 3.1.0 which output more information about the cost calculation.

This could be for e.g. the following cases:
- user not logged in
- no filament values in meta file 
- ...

Please test the new version, maybe we get a clear view why the costs not shown.

jason-a-alexander, 2020-10-08:
I installed the new version, and it's now working as expected.

closed 2020-10-08 by jason-a-alexander