OllisGit/OctoPrint-CostEstimation #9: No cost estimation when printing from Cura

mesiq, 2021-01-17:
When I'm printing directly from _Cura_ with _OctoPrint_ plugin, I have message **no filament from meta** in _Cost:_ field.

I'm using newest version of _CostEstimation_ with _SpoolManager_ (same issue was with _FilamentManager_ plugin). Should I configure Cura to pass any values or is it possible plugin will take data based on currently selected filament from _SpoolManager_?

_Note: CostEstimation works perfectly when initiating print from OctoPrint._

OllisGit, 2021-01-28:
Hi @mesiq ,

the message appears if OctoPrint is not able to detect the filament-properties like length, volumne. 
After sending the file from cure to op, is the print automatically started or do you press the print button on octoprint-side?

My current assumption is, that the API (with cura) use doesn't fire the event for calculating the meta data..

Jorael57, 2021-02-16:
Hi here,
Same Issue from PrusaSlicer with start print from the slicer.

mcmillanb, 2021-04-11:
Not sure if it helps, but if you change the settings in Filament Manager and check the box for 'Show dialog to confirm selected spools before starting/resuming the print job' every job prompts you to select which spool you are using and then it works. Direct from Cura doesn't. But I don't know how Cura would know what meta data to send to select the right spool

mesiq, 2021-04-11:
Thanks @OllisGit for picking up.

Sure Vita's API doesn't pass any data for filament. I was wondering if there is possibility to pass needed information or to start "currently used spool" in octopeont and that would be used for all prints by default.

I'm printing directly from Cura directly. Your plugin works perfectly when printing from octoprint.

OllisGit, 2021-05-22:
Hi,
I looked into the sourcecode of op and the problem is, that the needed metadata is missing after a direct print start.
You can find the metadata for each print in the upload folder, file called .metadata.json.

After the print is directly started from the slicer the metadata analyse is not executed, only after pressing pause or cancel.

You can test it: Start your print from cura, look into the ```.metadata.json``` file for your print job. The printjob is listed, but without analyse data (e.g. filament is needed for cost-calculation)
[image]
Stop your print and look again the analyse section was added.

I will ask the community: https://community.octoprint.org/t/metadata-analayse-not-executed-if-directly-print-from-slicer-with-autostart/34355

OllisGit, 2021-05-23:
Hi,

there is a solution: You need to change the analyse-feature in the Plugin-Settings, then the costs are shown during start and not at the end of a print.

[image]

Please test, if it works also with your slicer, I only tested Cura.

mesiq, 2021-05-25:
@OllisGit Tried that but without positive result, Octoprint still shows the same message at cost: no filament from meta

OllisGit, 2021-05-27:
@mesiq and what about the analyse data in the meta-file. When is it created?

OllisGit, 2022-01-16:
closed, no feedback.

closed 2022-01-16 by OllisGit

SapuSeven, 2022-02-04:
In my case this was indeed caused by the PrintTimeGenious plugin.
To resolve this, open PrintTimeGenious settings and enable "_Enable OctoPrint's built-in analyzer (slow and unnecessary)_". This allows the metadata file to be created.