OllisGit/OctoPrint-CostEstimation #10: Cannot set Default Filament Cost (without Filament or Spool Manager Plugin)

hweigel, 2021-02-06:
Hi Guys,

yeah, pretty much what the title says. I can't set the "default" Filament cost per gram/meter.

Octoprint 1.5.3 running on a Raspberry Pi 3b, CostEstimation version 3.2.

Tried cleanup, tried re-install, all to no avail. See screenshot for further details.

Thanks, cheers, stay safe, hope you can help a noob like me :-)

[image]

brandorf, 2021-03-16:
Also affected by this, I'm setting up a new printer.  It's estimating the cost using material cost from somewhere, but I can't see it in the UI to change it.

OllisGit, 2021-05-23:
Hi,

latest version 3.3.0 fixed this issue. Default values could be set anytime.

Please test and give feedback.

Thx, in advance
Olli

hweigel, 2021-05-24:
Hi @OllisGit, 

3.3.0 did it, thx for the fix, 's all good man!

Cheers
Helmut

[image]

closed 2021-05-24 by OllisGit