OllisGit/OctoPrint-CostEstimation #11: call cost estimation from other plugin

giloser, 2021-02-08:
Hi,

I work on the telegram plugin and I would like to be able to show a cost estimation when we select a file.
Before we were getting the options from the old cost plugin and doing a estimation but it's no more maintained and your seems way better.
Could it be possible to call to get the result of the estimation from an other plugin in python?

thanks a lot

adamboutcher, 2021-03-03:
I'd like to see support for Spool Manager...

OllisGit, 2021-04-04:
There are several ways to receive values, see
https://community.octoprint.org/t/opinions-on-data-exchange-between-plugins/18207/4

or you ask the plugin directly: 
```
plugin = self._plugin_manager.plugins[pluginKey]
neededData = plugin.implementation.giveMeTheData
```

So, the question is: which approach fits best for you...and what should I implement in CostEst-Plugin?

OllisGit, 2021-10-30:
Just for the records.
In the next version I added a public api method. Usage like described before:
```
plugin = self._plugin_manager.plugins["costestimation"] # needs to be improved, e.g. enabled
neededData = plugin.implementation.api_getCurrentCostsValues
```

closed 2021-10-30 by OllisGit