OllisGit/OctoPrint-CostEstimation #12: Plugins breaks upload and print from Prusaslicer (and presumably others)

WoLpH, 2021-04-12:
When this plugin is enabled I can't directly print from prusaslicer anymore. It will select the print but it won't start printing (i.e. I still need to press the print button in octoprint)

I should note that this is when printing to an Prusa MK3/S with MMU2S

OllisGit, 2021-04-13:
Hi @WoLpH,

please provide the octoprint.log. Maybe there is some error message in it.

WoLpH, 2021-04-13:
I believe it's related to being too fast :)
After restarting octoprint there's a few seconds that you can upload a print but it won't start the print. With this plugin enabled that time appears to be slightly increased.

It's pretty much a non-issue, I just noticed it because I was debugging an issue with a different plugin so I've restarted octoprint probably over a hundred times in the last few days ;)

**If you're experiencing this issue... wait a bit longer after starting octoprint for everything to finish**

closed 2021-04-13 by WoLpH