OllisGit/OctoPrint-CostEstimation #13: retail price

nowkillkennys, 2021-04-14:
it would be really cool to also have a retail price  i have some handles that i sell for £2.50 each and i tend to use it as a base line for all pricing

OllisGit, 2021-05-23:
Hi @nowkillkennys,

there are a couple of options how I can add a retail-price feature and I don't understand your example. So I need more informations from your side (maybe a screenshot/image).

E.g. new section in the plugin-settings "Retail" 
- textfield where you can enter a fixed price. This price is always added to the calculated price (e.g. used for packaging material/transport)
- textfield where you can enter a percentage which is added to the cost
- textfield "good friend bonus".....

OllisGit, 2022-01-16:
closed, no feedback

closed 2022-01-16 by OllisGit