OllisGit/OctoPrint-CostEstimation #15: [Suggestion] Cost estimation display in file list

tophattwaffle, 2021-10-04:
Display cost estimation in the file list for each print. Currently the file must be loaded to display the cost associated with the print. It would be nice to have the cost of each print displayed in the file list as well. This would allow for price estimations to be gathered without turning on / connecting a printer.

OllisGit, 2021-10-30:
Hi @tophattwaffle,
nice idea and I already started, but now I am blocked. 

It is not so easy to receice the spool selection event from the two plugins, so that I can repaint the whole file list after the user changed a single spool.
Maybe I found a good solution in the future.

[image]

plundrigan, 2023-11-06:
Could a version be released that will turn the feature on with the caveat that it does not work with Filament Manager or Spool Manager. I do not currently use the other plugins. Just useCost Estimation to get a ballpark of cost with power, standard filament cost, etc. I do agree that with what the OP said. This would save a lot of time not having to fire up a printer, or at min a pi to get a quick number.