OllisGit/OctoPrint-CostEstimation #24: Support for pulling filament info from Spoolman

Hillshum, 2024-06-08:
Spoolman is a web app for management filament. The OctoPrint-Spoolman plugin makes available information about filament cost that should be displayed by this plugin.

crazyduck, 2024-07-29:
+1 for spoolman

nikstohn, 2024-08-05:
+1 from here as well. Just switched from Spool Manager to Spoolman.
But I’m kinda afraid that this plugin got abandoned (again).

Hillshum, 2024-08-06:
I've working on implementing this actually. Hopefully this week I'll get it tested more and get a PR up. Still not sure how likely getting the PR approved will be.

nikstohn, 2024-08-06:
@Hillshum Awesome! But yeah, I highly doubt that a PR would even get noticed. Maybe we should open up an issue asking if this plugin is still maintained and after a while open up one on the OctoPrint plugin repository reporting its (very likely) abandonment.

Hillshum, 2024-08-10:
Well, I opened https://github.com/OllisGit/OctoPrint-CostEstimation/pull/26

Hillshum, 2024-11-21:
See https://github.com/Hillshum/OctoPrint-CostEstimation/pull/8

Will have this up shortly

Hillshum, 2024-11-25:
This is included in https://github.com/Hillshum/OctoPrint-CostEstimation/releases/tag/v3.6.0rc1 which is available in the pre-release channel that you can enable from the software update panel within your  OctoPrint install. Testing would be appreciated